PX4/pyulog #10: Decode error on pyulog streamed over wifi

jgoppert, 2016-12-02:
Here are the log files I colleted, some of them work, but the large one doesn't even after updating to master: 

https://drive.google.com/open?id=0B3wIEs1gcVahNlVydy0zazZ5bU0

```bash
ulog_info 001-2016-12-02-13-21-25-513.ulg 
Traceback (most recent call last):
  File "/home/jgoppert/anaconda3/bin/ulog_info", line 11, in <module>
    sys.exit(main())
  File "/home/jgoppert/anaconda3/lib/python3.5/site-packages/pyulog/info.py", line 25, in main
    ulog = ULog(ulog_file_name)
  File "/home/jgoppert/anaconda3/lib/python3.5/site-packages/pyulog/core.py", line 265, in __init__
    self.loadFile(file_name, message_name_filter_list)
  File "/home/jgoppert/anaconda3/lib/python3.5/site-packages/pyulog/core.py", line 276, in loadFile
    self.read_file_data(message_name_filter_list)
  File "/home/jgoppert/anaconda3/lib/python3.5/site-packages/pyulog/core.py", line 328, in read_file_data
    self.message_formats)
  File "/home/jgoppert/anaconda3/lib/python3.5/site-packages/pyulog/core.py", line 164, in __init__
    self.message_name = parseString(data[3:])
  File "/home/jgoppert/anaconda3/lib/python3.5/site-packages/pyulog/core.py", line 14, in parseString
    return str(cstr, 'ascii')
```

bkueng, 2016-12-02:
Did you use QGC? There is a very recent fix, https://github.com/mavlink/qgroundcontrol/pull/4242, do you have this?

jgoppert, 2016-12-02:
I was using daily @d1cc23

jgoppert, 2016-12-03:
I haven't been able to reproduce this so will close it. It must have been due to an early device power off or something.

closed 2016-12-03 by jgoppert

bkueng, 2016-12-05:
Your QGC version is before the merge of 4242, so that would explain it as well.