PX4/pyulog #13: Attitude transformation base: body / inertial frame

LorenzMeier, 2017-01-02:
@bkueng @jgoppert We should update the routine here to allow different references:
https://github.com/PX4/pyulog/blob/b68e3a6b42a9ac6f8b577885cf343639d0641b9b/pyulog/px4.py#L76

@MaEtUgR Our flight logs show that the rate tracking is excellent but the angle tracking is off, depending on the vehicle yaw. I think there is something incorrect in the way we report roll and pitch set points currently. Example from this log:

http://logs.px4.io/plot_app?log=e55ed4c2-09db-416b-b622-98c3a18c948e

The rate tracking is excellent the angle tracking is horrible. The vehicle would have flown away given those tracking errors, so its just the reporting, not the actual flight performance which is wrong.

[image]

[image]

If you correlate it with the yaw plot it becomes obvious that there is a missing transformation from inertial to body here. And it is interesting that the yaw angle tracking seems to be off but that's probably not what happened. This is using #6213. Is the removal of the attitude setpoint publications maybe uncovering this?

LorenzMeier, 2017-01-02:
One of the two clearly has the wrong frame of reference and/or we're missing one transformation.

bkueng, 2017-01-03:
I did some investigations, with the following observations so far:
- FlightPlot shows the same graphs:
[image]
- Plotting and converting from the quaternion in FlightPlot shows that `vehicle_attitude_setpoint.roll_body` and `q_d` match.
- Comparing with sdlog2: it does the same conversion from `vehicle_attitude.q` to get the roll, pitch & yaw. And for the setpoint it uses the `roll_body` which matches the quaternion data `q_d`.

Also a SITL mission does not show the problem: http://logs.px4.io/plot_app?log=7df09d3a-bc18-4fee-a9f8-d7d65cb0a9d5
I need to look closer what happens on the Firmware side.

LorenzMeier, 2021-03-12:
Closing as stale.

closed 2021-03-12 by LorenzMeier