PX4/pyulog #23: ulog2kml fails with missing module simplekml

hamishwillee, 2018-05-17:
Library is missing dependency (all other scripts work). This is on Ubuntu 16.04 LTS, Python 2.7 default, installed using either method. 

```
ulog2kml -h
Traceback (most recent call last):
  File "/usr/local/bin/ulog2kml", line 7, in <module>
    from pyulog.ulog2kml import main
  File "/usr/local/lib/python2.7/dist-packages/pyulog/ulog2kml.py", line 9, in <module>
    import simplekml # pylint: disable=import-error
ImportError: No module named simplekml
```

bkueng, 2018-05-17:
simplekml is not installed on purpose to reduce the number of dependencies, especially since it's now installed on all dev machines.
if you feel strongly, I'll add the dependency, otherwise we can add a more informative message.

hamishwillee, 2018-05-17:
Well, it wasn't installed on my test machine for this, so not *all* test machines. I tend to think that if something is needed the dependency is added to the PyPi definition. If you don't wan to do this, then IMO we just need to add docs for the setup to state how it is installed. Is it just `pip install simplekml`?

bkueng, 2018-05-17:
Yes, it's `pip install simplekml`

bkueng, 2018-06-08:
Closing, since it's documented in the README now.

closed 2018-06-08 by bkueng