PX4/pyulog #26: Feature request - Add Timestamp when parameters changes during flight

BazookaJoe1900, 2018-06-08:
When parameter is being change during flight there is only list of the values, but not their timestamp of being changed.
For PID tuning and other peruses its important to know the time.
I seen that the 'changed_parameters' array has a timestamp value, but its not correct when I have tried to print it.

Adding ulog_params arg to print the timestamp can help alot

tstastny, 2018-07-26:
This would be super useful -- I was also trying to get this to work recently, but when printing the timestamps I only get the same "t" value for all "changed_parameters" (see [pyulog/params.py#L48](https://github.com/PX4/pyulog/blob/master/pyulog/params.py#L48)). @BazookaJoe1900 was this also what you experienced?

@bkueng do you possibly have any insights on the best way to accomplish this? It must be possible, as the changed params are shown at the right timestamp on the review.px4.io

bkueng, 2018-07-28:
I fixed the timestamps in https://github.com/PX4/pyulog/commit/cb14d687289a62cb824bec0ed14374b0422ebcbb. Can you create a PR with the format you would like to see?

tstastny, 2018-07-28:
> I fixed the timestamps in cb14d68. Can you create a PR with the format you would like to see?

@bkueng thanks! I've made a PR with a proposed format, see #29

PR #29 (2019-04-29): params: output changed parameter timestamps in csv
Outputs changed parameters and their timestamps in csv. This is just one way to format this -- open to other ideas. "0" is used as the timestamp for the initial parameter.

resolves #26 

Output in the csv formatted like this:

| column 1      | column 2      | column 3      | column 4      |
| ------------- | ------------- | ------------- | ------------- |
| PARAM 1       | init val      | changed val 1 | changed val 2 |
| timestamp     | 0             | timestamp 1   | timestamp 2   |
| PARAM 2       | init val      | changed val 1 | changed [... cut]
closed 2019-04-29 by bkueng