PX4/pyulog #28: Conversion failed when I used ulog2kml

MyOnlyOnlySunshine, 2018-07-11:
Conversion failed when I used ulog2kml，but successful when i use ulog2csv

bkueng, 2018-07-11:
What do you mean by fails? Can you provide the output? Did you install `simplekml`?

MyOnlyOnlySunshine, 2018-07-11:
yeah, I installed simplekml  before use ulog2kml , but have no kml file output.  My device is mac .

sampathkumarch, 2019-05-01:
How convert ulg 2 Excel 4 csv

sampathkumarch, 2019-05-01:
How to convert ulg to kml please tell me step by step process

bkueng, 2019-05-03:
For kml, see here: https://github.com/PX4/pyulog#convert-ulog-to-kml-files-ulog2kml, csv is also documented there.

sampathkumarch, 2019-05-03:
but i have got errors how to resolve them
 ERROR: Could not find a version that satisfies the requirement pylog (from
versions: none)
ERROR: No matching distribution found for pylog

bkueng, 2019-05-03:
Please provide the exact commands and ouput you get

sampathkumarch, 2019-05-03:
ok  but now i am getting another error          :python: can't o  pen file
'install': [Errno 2] No such file or directory

 please tell me commands which commands i dont about this knowledge thats
why am asking please tell me ans

sampathkumarch, 2019-05-03:
Okk I got  it final output . thanks for given a reply

closed 2019-05-04 by kd0aij