PX4/pyulog #33: Shift GPS data to make log location anonymous

ryanjAA, 2018-10-08:
There are quite a lot of scenarios where log data needs to be made somewhat anonymous prior to being able to being shared. After some conversations with @dagar, this can be a bit more tricky than initially thought since everything from gps, global position, home position, mission items, setpoints to even mag declination give a pretty clear indication of the location of the flight. There are just cases where substantial amounts of flight data can be used for analysis but often reporting where those flight took place isn't allowable, for whatever reason. 

Initial thoughts would be to create the ability to have a random offset of all location data and then hold that offset for all coordinates in the log, that way everything is related and will still follow course. for analysis. This is obviously very preliminary but here to get the conversation started.

LorenzMeier, 2018-10-08:
@bkueng and I looked at that today as well. We definitely would like to enable that.

ryanjAA, 2018-10-08:
That would be good. We were given BVLOS clearance last week but the location is not something we can report openly so cases like that make this something that is a newly present issue. We want to be able to share test flights but first things first.

LorenzMeier, 2021-03-12:
Closing as stale.

closed 2021-03-12 by LorenzMeier

junwoo091400, 2023-03-27:
Hi! Just as a heads up, I have built this functionality, and can be found here:

https://github.com/junwoo091400/Anonymize_ULog_Data

ryanjAA, 2023-03-27:
Awesome!! Thanks so much!