PX4/pyulog #34: unable to load px4 logs

chataign, 2018-10-26:
I am running offboard missions and generating logs using the current master of px4/Firmware (nuttx_px4fmu-v2_default config). However I am unable to load those logs into pyFlightAnalysis as pyulog (v0.6.0) fails to load the files. 

```
Traceback (most recent call last):
  File "ulog_test.py", line 6, in <module>
    log_data = ULog(log_file_name).data_list
  File "/usr/local/lib/python2.7/dist-packages/pyulog/core.py", line 108, in __init__
    self._load_file(log_file, message_name_filter_list)
  File "/usr/local/lib/python2.7/dist-packages/pyulog/core.py", line 428, in _load_file
    self._read_file_header()
  File "/usr/local/lib/python2.7/dist-packages/pyulog/core.py", line 450, in _read_file_header
    raise Exception("Invalid file format (Failed to parse header)")
Exception: Invalid file format (Failed to parse header)
```

bkueng, 2018-10-26:
Can you provide the log file? How did you retrieve the file?

chataign, 2018-10-26:
Hi, I retrieve the logs using the file manager in QGroundControl.

Here's an example log file: [log_370_2018-10-25-17-25-56](https://github.com/PX4/pyulog/files/2519409/log_370_2018-10-25-17-25-56.ulg.zip)

bkueng, 2018-10-26:
Thanks. The file contains only zeros. Can you try:
- take out the SD card and copy the file directly?
- reformat the SD card and/or try a different one?

Which board are you using?

chataign, 2018-10-26:
I just formatted the SD card and generated the following log: 
[14_55_37.ulg.zip](https://github.com/PX4/pyulog/files/2519726/14_55_37.ulg.zip)

There is also a log in a 'sess001' folder:
[log001.ulg.zip](https://github.com/PX4/pyulog/files/2519728/log001.ulg.zip)

I copied them directly from the SD card I tried to open both with pyFlightAnalysis but no success.

Thanks,

chataign, 2018-10-26:
And we are using the original Pixhawk board (v1 I believe)
[image]

bkueng, 2018-10-27:
These logs are both good. Did you get the same error?

bkueng, 2018-11-05:
Closing this then.

closed 2018-11-05 by bkueng