PX4/pyulog #38: String decoding errors on python3

devbharat, 2019-05-09:
Currently python3 implementation of Ulog decodes to ascii and throws exception on decoding errors. That can be an issue in case there is a corrupt byte in the stream or non-ascii bytes. What would be your thought on switching to utf-8, and setting str() error handling to ignore?

bkueng, 2019-05-09:
I prefer the exception because it makes it clear the log is corrupt.
If you want to make it an option, that works for me as well.

devbharat, 2019-05-09:
Thanks, that sounds good. I'll make a PR.

devbharat, 2019-05-09:
Just to be clear, the defaut behaviour currently is different for python 2 and python 3. Python2 silently ignores and python3 fails over. I think the default behaviour should report the error and print warning but not failover, keeping pyulor robust to external errors.

bkueng, 2019-05-14:
> I think the default behaviour should report the error and print warning but not failover, keeping pyulog robust to external errors.

There's a trade-off: if you have parsing errors, you cannot trust the data. Can you make it a parameter `disable_parser_exceptions` to ULog(), and if True, set `self._file_corrupt` instead of raising the exception?

devbharat, 2019-05-14:
> There's a trade-off: if you have parsing errors, you cannot trust the data.

While that's true, often log analysis comes down to recovering as much logged info from a logfile as possible - especially for crash logs. Only then you'd consider the matter of 'trustability' of the data. I can't imagine a scenario where a user expects a log parsing pipeline to stop midway and discard rest of the logged data because it encountered unexpected bytes (especially true if the corrupt bytes are not message identifiers, but string payload of message packets!).

Still unclear points to me in discussion are difference in python2 and python3 behavior, and whether ascii should be the default or utf-8.

> Can you make it a parameter disable_parser_exceptions to ULog(), and if True, set self._file_corrupt instead of raising the exception?

Sounds like a good idea.

devbharat, 2019-05-14:
Also note https://github.com/PX4/pyulog/issues/40

bkueng, 2019-05-15:
As I wrote I'm fine with making this configurable (the default can be either way). What's important to me is that we have the reporting of exceptions, both in Flight Review and the scripts.

> Still unclear points to me in discussion are difference in python2 and python3 behavior, and whether ascii should be the default or utf-8.

Preferably utf-8.

PR #42 (2019-06-08): Parametrize string parsing error handling
Fixes #38 

@bkueng I do not much like setting the static variable in the __init__ fn, but the alternative I see is passing parameters everywhere till they reach _parse_string() which is ugly as well.
closed 2019-06-08 by LorenzMeier