PX4/pyulog #39: Logging messages with a 'Tag'

devbharat, 2019-05-09:
The current `MSG_TYPE_LOGGING` packet holds severity, timestamp and a generic message string. It'd be helpful if we could extend this to add a 'tag' string to signify the source of the generated message (instead of encoding it into the message itself). Then ulog could parse messages from each source into a dictionary with the tag as the key. Alternatively, we could create a new message type with the tag included. Which would be a preferred option?

bkueng, 2019-05-14:
What sources do you have in mind, and what is the use-case?

devbharat, 2019-05-14:
Say an onboard computer running multiple processes that also generate their own logs. A simple approach would be to have separate logfiles from all of those processes, and a separate streamed logfile received from PX4, but a better more integrated approach would be to queue up ulog formatted messages from all the processes, one of them being the PX4 log-streaming sink, and flush them into a single ulog file. 

In my opinion it greatly simplifies data handling of logs from users, and facilitates a single integrated postprocessing pipeline.

bkueng, 2019-05-15:
That makes sense. Generally we cannot simply extend existing messages, adding a new one would be easier.
Or you could use `message_info_multiple`, with a different key per component, but it lacks the log level.

devbharat, 2019-05-15:
> Generally we cannot simply extend existing messages, adding a new one would be easier.

I made a prototype `log_message_obc` exactly same as log_message but with another uint8_t called `tag` next to severity. The message length I still kept at 128 chars, but am thinking about increasing it to 256. So this generally sounds good to you?

bkueng, 2019-05-16:
So tag is more like an enum? I'd use `uint16_t` then.
The current message length is 128 to limit stack size usage. This is not a hard limit though.

devbharat, 2019-05-16:
> So tag is more like an enum?

I thought it better to switch to that, and leave the mapping to string names to the frontend.

PR #43 (2019-06-07): Feature/add tagged logging
Fixes #39 
Add tagged logging with uint16_t tag attribute in every msg.
closed 2019-06-07 by bkueng