PX4/pyulog #52: ulog_params doesn't show parameter values

George-avy, 2020-06-15:
Hi everyone! I have this weird issue: When I use ``ulog_params``, I get only the parameter names, not parameter values.

```bash
$ ulog_params mylogfile.ulg
ASPD_BETA_GATE
ASPD_BETA_NOISE
ASPD_DO_CHECKS
ASPD_FALLBACK
ASPD_FS_INNOV
ASPD_FS_INTEG
ASPD_FS_T1
ASPD_FS_T2
ASPD_PRIMARY
ASPD_SCALE
```
I'm using pip-installed version 0.8.0:
```bash
$ pip install --upgrade pyulog
Defaulting to user installation because normal site-packages is not writeable
Collecting pyulog
  Downloading pyulog-0.8.0.tar.gz (36 kB)
Requirement already satisfied, skipping upgrade: numpy in /home/george/.local/lib/python2.7/site-packages (from pyulog) (1.16.5)
Building wheels for collected packages: pyulog
  Building wheel for pyulog (setup.py) ... done
  Created wheel for pyulog: filename=pyulog-0.8.0-py2-none-any.whl size=22879 sha256=bc03f68916ee7ff70a232e0e5bead97fa4d4c0c86f58e7a4b1c2f51e75ed6d73
  Stored in directory: /home/george/.cache/pip/wheels/58/cb/81/e3e83fba56877bceafc1a6dd4c77f6bc659bde5a445e6e6e96
Successfully built pyulog
Installing collected packages: pyulog
Successfully installed pyulog-0.8.0
WARNING: You are using pip version 20.0.2; however, version 20.1.1 is available.
You should consider upgrading via the '/usr/bin/python2 -m pip install --upgrade pip' command.
```

bkueng, 2020-06-16:
Hi
Can you provide the log?

George-avy, 2020-06-16:
Err, actually I can't because today every log I have works just fine. Weird...
Welp, nothing to see here anymore. Sorry for bothering you.

closed 2020-06-16 by George-avy

bkueng, 2020-06-16:
No worries