PX4/pyulog #63: not able to fin binary installed

RaffoBags, 2021-11-01:
Hi all, sorry I am sure this is a dumb question;
I use ubuntu 18.04 and I installed the package as indicated. the result is this:

user1@user1-VirtualBox:~$ python3.7 -m pip install pyulog
Collecting pyulog
  Using cached https://files.pythonhosted.org/packages/f4/f8/00495d753b987fe228f3eb1b81c32a7bab5339627f52e30a4b887af7bdb0/pyulog-0.9.0-py3-none-any.whl
Collecting numpy (from pyulog)
  Using cached https://files.pythonhosted.org/packages/b0/27/86c9c52a335695fabc79514a196abd2d703ff8b089ba3207f069b63115ce/numpy-1.21.3-cp37-cp37m-manylinux_2_5_x86_64.manylinux1_x86_64.whl
Installing collected packages: numpy, pyulog
Successfully installed numpy-1.21.3 pyulog-0.9.0

Now I don't know where to find the binary of ulog2cvs that is required in the .m file
% REPLACE this with the location of your installed ulog2csv binary
handles.ulog2csv = '???';

Sorry again, hope you can help me

bkueng, 2021-11-02:
Is it found when you enter `ulog2csv` on the shell? You can use `which ulog2csv` to get the path.

RaffoBags, 2021-11-02:
hello, thanks for your quick reply. If I type which ulog2cvs it does nothing: 
user1@user1-VirtualBox:~$ which ulog2cvs
user1@user1-VirtualBox:~$ 

and if I type ulog2cvs it says command not found
user1@user1-VirtualBox:~$ ulog2cvs
ulog2cvs: command not found
user1@user1-VirtualBox:~$ 

am I doing something wrong?

Thanks a lot

bkueng, 2021-11-03:
You need to find where `python3.7 -m pip install pyulog` installs the scripts. It's typically `~/.local/bin`, but depends on the system. If you do `ls ~/.local/bin` and it's there, you need to add this path to [`$PATH`](https://opensource.com/article/17/6/set-path-linux).
Otherwise you can try explicitly installing as user: `python3.7 -m pip install --user pyulog`.

RaffoBags, 2021-11-03:
Hello, thanks for your message.

I do see ulog2cvs inside ~/.local/bin. thank you for that.
Now, the matlab script that I am using, which is the one that needs the path to that binary, generates this command to be executed in order to read the file. 
/home/user1/.local/bin/ulog2cvs -o/home/user1/Documents/Temporary/100722 /home/user1/Documents/log_1_2019-8-22-18-16-40.ulg

This text is saved inside a variable 'sys_cmd', and the matlab command to execute that text ion the shell is
 [status,result] = system(sys_cmd).

What I get out of this is the following message:

 status =

   127

result =

    '/bin/bash: /home/user1/.local/bin/ulog2cvs: No such file or directory
     '

Which is the same thing that I get if i copy the text in sys_cmd and paste in the shell.
So either the command that generates has a wrong syntax or something else.
Do you know what would be the correct way to execute that ulog2cvs program?

Thanks a lot for your help, appreciated!

bkueng, 2021-11-04:
Does it work after you do `chmod +x /home/user1/.local/bin/ulog2cvs`?
What is the output of `ls -l /home/user1/.local/bin/`?

RaffoBags, 2021-11-04:
Hello,
thanks for your message.

if I type chmod +x /home/user1/.local/bin/ulog2cvs on the shell I get:

user1@user1-VirtualBox:~$ chmod +x /home/user1/.local/bin/ulog2cvs
chmod: cannot access '/home/user1/.local/bin/ulog2cvs': No such file or directory

the output of ls -l /home/user1/.local/bin/  shows a lot of things and in the last few lines I see ulog2cvs as reported below

-rwxrwxr-x 1 user1 user1   220 Nov  3 10:00 ulog2csv
-rwxrwxr-x 1 user1 user1   220 Nov  3 10:00 ulog2kml
-rwxrwxr-x 1 user1 user1   228 Nov  3 10:00 ulog_extract_gps_dump
-rwxrwxr-x 1 user1 user1   216 Nov  3 10:00 ulog_info
-rwxrwxr-x 1 user1 user1   220 Nov  3 10:00 ulog_messages
-rwxrwxr-x 1 user1 user1   218 Nov  3 10:00 ulog_params
-rwxrwxr-x 1 user1 user1   208 Oct 20 08:51 wheel

Thnka a lot for your help

bkueng, 2021-11-05:
I'm blind. It's a simple typo: `ulog2cvs` -> `ulog2csv` (comma-separated-values)

RaffoBags, 2021-11-05:
doh! that was silly, sorry about that!!
Anyway, after using the correct name it worked fine.
However I want to leave a note for whoever will arrive at this page because of the same initial issue:
1) in the ulogviewer.m script change line 59 with handles.ulog2csv = '/home/user1/.local/bin/ulog2csv'; if you're not sure which is the path to that file use ''which ulog2csv"

2)For me, the program does not create the temp folder that is defined at line 93. 
I solved by creating a folder and relative subfolder myself myself like /home/user1/Documents/Temporary/092213
and replaced line 93 with  temp_dir = [fileparts(mfilename('fullpath')) filesep 'Temporary' filesep datestr(now,'092213')];

Thanks a lot bkueng for your help and patience!!!

closed 2021-11-08 by bkueng