PX4/pyulog #84: px4.add_roll_pitch_yaw doesn't work with DatabaseULog.get_dataset

hermankolden, 2023-09-02:
When I call px4.add_roll_pitch_yaw(), I expect e.g. 'vehicle_attitude.roll' to be retrievable from uog.get_dataset('vehicle_attitude'). However, since the former modifies the ULog object directly and the latter gets the dataset from the DB, the 'roll' field is not found.

hermankolden, 2023-09-02:
IMO we should refactor this to avoid direct object modifications. Maybe e.g. adding preprocessors in DatabaseULog.save(), so that roll/pitch/yaw is always added.

bkueng, 2023-09-04:
I would rather go the other way and make sure that post processing works, otherwise there's inconsistent behavior between the file and DB, in terms of what data is stored.
The problems I see are:
- `get_dataset` always loads from DB when called repeatedly for the same dataset
- potentially lazy loading and `add_roll_pitch_yaw` does not call `get_dataset` to fill in the data

That would still allow you to do the modifications before calling `save()`.

closed 2024-01-22 by bkueng