PX4/pyulog #92: pyulog represents

BWStearns, 2024-02-07:
pyulog represents `char[x]` fields as `x` number of `str` fields. It seems less surprising to represent them all as one string instead of having to have a postprocess that concats them later. In the docs it's described as a char[127] which would more intuitively map to a string than 127 char values each with their own key (or in this case column).

[image]

```
# A logging message, output with PX4_{WARN,ERR,INFO}

uint64 timestamp		# time since system start (microseconds)

uint8 severity # log level (same as in the linux kernel, starting with 0)
char[127] text

uint8 ORB_QUEUE_LENGTH = 4
```
https://docs.px4.io/main/en/msg_docs/LogMessage.html

bkueng, 2024-02-09:
We can add a helper to do the conversion. It makes sense, but I'd like to avoid breaking API changes or special handling in the parser for this.

BWStearns, 2024-02-09:
A helper would be great.

bkueng, 2024-02-12:
Can you create a PR?

BWStearns, 2024-02-12:
I can. I just need to figure out where to put the changes. Got any pointers off the top of your head for where would make sense to put it?

bkueng, 2024-02-13:
You could add it to the `Data` class: https://github.com/PX4/pyulog/blob/main/pyulog/core.py#L487

bkueng, 2025-05-06:
@BWStearns were you able to implement it?

closed 2026-10-27 by bkueng