PX4/pyulog #107: Need help with updated ULG file format error

RobPritt, 2025-02-09:
Hi, I have been using your ULG2CSV to convert a drone's ULG file to CSV. However they apparently did something to their ULG files in latest drone update and ULG2CSV now chokes on them and throws an error along the lines of UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 9: invalid start byte
The location varies for each ulg file.. I had previously been able to convert dozens of their ULG files without a problem.
Would you be willing to take a look at the ULG files if I send them to you?

bkueng, 2025-02-11:
Can you try with `ulog2csv -i <file.ulg>`. This makes it ignore string parsing exceptions.

RobPritt, 2025-02-11:
Thankyou. That worked. I added “-i” to my current line..

 ulog2csv -i <file.ulg> –o <output folder>

Would you like me to send you some ULG files with the error to debug? I would be worried that ignoring certain things will cause loss of some data.

Rob
From: Beat Küng 
Sent: Tuesday, February 11, 2025 12:16 AM
To: PX4/pyulog 
Cc: RobPritt ; Author 
Subject: Re: [PX4/pyulog] Need help with updated ULG file format error (Issue #107)

Can you try with ulog2csv -i <file.ulg>. This makes it ignore string parsing exceptions.

bkueng, 2025-02-12:
Yes I can have a look. I agree with you, that's why the flag is not set by default.
There's one case where I've seen this happen without there being a memory corruption. It's when the NuttX boot console outputs some special characters (which end up in the log).

RobPritt, 2025-02-13:
Did you get my ULG files OK?

Rob

From: Beat Küng 
Sent: Tuesday, February 11, 2025 11:19 PM
To: PX4/pyulog 
Cc: RobPritt ; Author 
Subject: Re: [PX4/pyulog] Need help with updated ULG file format error (Issue #107)

Yes I can have a look. I agree with you, that's why the flag is not set by default.
There's one case where I've seen this happen without there being a memory corruption. It's when the NuttX boot console outputs some special characters (which end up in the log).

bkueng, 2025-02-14:
No, I did not see anything

RobPritt, 2025-02-19:
I sent you the ULG files last week. I was expecting to hear back from you.

bkueng, 2025-02-20:
Sorry I did not receive anything, I also checked the spam folder. Can you PM me on PX4 discord?

RobPritt, 2025-02-20:
I have no idea what PX4 discord is.

I emailed you several times. Shouldn’t have to be so difficult!

From: Beat Küng 
Sent: Wednesday, February 19, 2025 10:39 PM
To: PX4/pyulog 
Cc: RobPritt ; Author 
Subject: Re: [PX4/pyulog] Need help with updated ULG file format error (Issue #107)

Sorry I did not receive anything, I also checked the spam folder. Can you PM me on PX4 discord?

—
Sorry I did not receive anything, I also checked the spam folder. Can you PM me on PX4 discord?

bkueng, 2025-02-25:
I'm open to other suggestions.

closed 2025-05-06 by bkueng